LoupHC/controleur-CAPE #92: PCB - schematics

LoupHC, 2019-02-24:
Salut Yoan!
Quelques petits commentaires, j'attend ton ok pour faire les modifications dans les schematics:
- pour la protection des inputs, je remplacerais les ptc par des résistances 220ohm, je ne trouve rien en TH qui face la meme job que dans leur circuit :  
https://www.rugged-circuits.com/10-ways-to-destroy-an-arduino/
avec une résistance, on n'aura pas l'effet fusible, mais au moins on est sûr de rester en bas de 20mA meme avec des spikes de 50V
- pour l'autre ptc avant le driver de relais, les transistors vont fuser avant elle, donc je ne pense pas que ça vaille la peine non plus... 
- les deux ptc avant les modules dc-dc, quelle est leur fonction?
- est-ce qu'on met vraiment un fusible sur le 24v en plus du circuit de protection contre les surtension du PS?
- quelle est la fonction de la switch pour changer de modèle d'anémomètre?
- il n'y a pas de fusibles en série avec les moteurs, est-ce que c'est voulu?

yoangp, 2019-02-25:
- Pour les inputs, j'avais trouvé le PTC suivant: https://www.digikey.ca/product-detail/en/amphenol-advanced-sensors/YQD100N1000/235-1259-ND/2324591
C'est le seul qui semblait faire environ la job, résistance de base de 1kohm, courant trip 30mA. J'ai vu que 1kohm de résistance pouvait être correct sur les inputs.
Tu peux regarder ça de ton bord, mais si on laisse faire le ptc et qu'on met juste une résistance, alors je m'assurerait qu'elle est assez élevé pour qu'on moins 12V en entrée ne fasse pas buster le courant max permi sur les inputs du Atmega.

- ok pour enlever le ptc avant le driver de relais

- Les deux ptc avant les modules dc-dc c'était pour limiter le courant qui rentre dedans si ils font défaut. Mais je sais pas si c'est vraiment nécessaire. Ils sont vraiment pas mal bien protégés ces nouveaux dc-dc. Faudrait voir dans les datasheets s'ils recommandent un fusible en avant, avec l'ancien module que l'avais sélectionné c'était recommandé. Sinon, regarder sur le net si c'est une bonne pratique.. Si pas nécessaire, on les enlève.

- Fusible sur le 24v; je pense que c'est important, car c'est la première protection si le DC-DC converter 24V fait défaut ou laisse passer un spike de voltage relativement gros. Je ne sais pas vraiment à quel point le surge protector élimine tout.. Je pense que ça empêche de détruire le power supply, mais c'est peut-être pas suffisant pour que la sortie du DC-DC reste clean pendant la surtension...

- * switch pour changer de modèle d'anémomètre; voir le issue sur l'anémomètre, mais en gros c,est pour changer de modèle d'anémomètre entre Davis et Sparkfun. Davis est construit avec potentiomètre en diviseur de tension (l'output sur pin du milieu, GND et +5V aux extrémités). Sparkfun construit comme résistance variable; donc ça prend une résistsance sur le circuit pour faire le diviseur de tension. Mais on pourrait peut-être juste brancher les 2 comme une simple résistance variable, donc pas besoin de switch (mais ajustement dans le code car les valeurs de résistances et donc de tension analog ne seront pas pareilles)... Il faudrait juste regarder les valeurs de résistances dans 2 modèles et voir si ça va être relativement similaire dans les valeurs de tension lues. 

- Pas de fusibles moteurs, je pense qu'on devrait vraiment mettre le courant max dans le code via les acs723. Si court-circuit moteur, ça arrête à 6A, ce qui donne pas beaucoup plus que les 100W max des moteurs.

Ciao!

yoangp, 2019-03-03:
- Je propose de remplacer les pct sur les inputs par des résistances de 1K, si les inputs sont à low (0V) et qu'il y a 24V en entrée ça fait 24mA qui passe dans la pin d'input.

- J'ai enlevé les ptc sur les DC-DC converters, la fuse de 1,5A devrait faire la job s'il y a un short dans la circuiterie d'input d'un des deux dc-dc conveter 5V et 12V.