LoupHC/controleur-CAPE #94: Circuit d'alarme

LoupHC, 2019-03-31:
Nouveau circuit (un peu) plus compliqué pour l'alarme. Il comprend deux nouveaux features : 
- interruption du buzzer interne grâce à un pushbutton plutot quune switch à bascule.
- une led reste allumée tant que le problème persiste
- une fois que le problème est détecté et corrigé, le buzzer est ré-armé pour sonner au prochain problème

Le circuit comprend aussi un connecteur pour brancher une alarme externe basique avec alarme+switch_sirène
[image]

Description du fonctionnement : 
1. Un problème (court-circuit, température trop élevée) relâche le relais qui ferme le circuit d'alarme
2. le condensateur 22uF génère un pulse unique de quelques mili-secondes
3. le pulse "arme" le circuit flip-flop pour le mettre en mode "buzzer"
4. une pression sur le pushbutton met le flip-flop en mode "muet"
5. Tant que la tension ne baisse pas au pôle positif du condensateur, le circuit reste bloqué en mode muet. Quand l'alarme est désactivée, le condensateur se décharge et est donc prêt à générer un deuxième pulse la prochaine fois que le relais se relâche.
6. la LED reste allumée tant que le relais est relâché

yoangp, 2019-05-06:
Cool! Est-ce que tu as testé ton circuit sur bread-borad ou bien on y va cowboy et on teste sur le pcb prototype?

yoangp, 2019-05-07:
Allo Loup, est-ce que tu as updaté une version du schéma avec cet ajout pour l'alarme? Je le trouve pas sur Dropbox ou dans Github. Sinon, est-ce que tu voulais que j'intègre cela au schéma que j'ai fait? Merci!

LoupHC, 2019-05-10:
Ah oui justement! J'ai modifié le circuit et ce coup-ci il marche super bien
[image]

LoupHC, 2019-05-10:
oups!! D33 est à l'envers!

yoangp, 2019-05-19:
Ou as tu trouvé les transistors S9013? Il n'y en a pas chez Digikey...

yoangp, 2019-05-19:
Ça aurait été cute un IC flip-flop tout en un..

LoupHC, 2019-05-19:
Hmm faudrait trouver un transistor EBC de remplacement...
je peux essayer de trouver ça...
Sinon ce site suggère qu'on peut utiliser des darlingtons pour un flip flop, en modifiant les valeurs des résistances... si c'est le cas on pourrait réutiliser le mêm ULN2075 du circuit de drive des relais
http://www.techlib.com/electronics/flipflopflash.htm
Je peux essayer de breadboarder ça vite fait cette semaine!

LoupHC, 2019-05-19:
avec des darlington en plus on pourrait brancher une alarme mega forte sur le même circuit dans soucis

yoangp, 2019-05-22:
Ajout d'une mesure du potentiel de la batterie de l'alarme pour vérifier si la batterie est trop déchargée ou déconnectée. Infos sur le diviseur de potentiel à ajouter : https://forum.arduino.cc/index.php?topic=99318.0

yoangp, 2019-05-22:
Laisse faire pour refaire le circuit, je vais essayer de trouver un équivalent au transistor: https://alltransistors.com/crsearch.php?mat=Si&struct=NPN&pc=0.3&ucb=40&uce=25&ueb=5&ic=0.5&tj=150&ft=150&cc=0&hfe=120&caps=SOT23&r1=0&r2=0&r1r2=0

yoangp, 2019-05-22:
Bah! J'ai pas assez d'infos sur le transistor que tu as utilisé! C'est quoi le numéro exact du transistor que tu as pris? S9013... ? Sinon les valeurs de résistances et tout ce qui va autour doit changer. C'est plus rapide si on change rien du tout à part le numéro de transistor.

yoangp, 2019-06-01:
Loup! Si t'as quelques minutes pour me donner le numéro complet du S9013 que tu as utilisé ça serait super! J'ai besoin de cela pour vérifier la datasheet et trouver le bon équivalent chez digikey.. Merci ;-)

LoupHC, 2019-06-01:
Je n'ai rien d'autre que S9013 écris... mais n'importe quel transistor avec le pinout CBE fera la job, le max de courant C-E  correspont au courant max qui passerait dans une sirène, en haut de 500mA on est safe je pense.
Pour la tension C-E max le s9013 est raté à 25V, je pense qu'on pourrait viser quelque chose de plus élevé, par sécurité, mais en même temps c'est pas un circuit très à risque.

LoupHC, 2019-06-01:
Ça devrait le faire ça : https://www.digikey.ca/product-detail/en/central-semiconductor-corp/2N5551/2N5551CS-ND/4806921