LoupHC/controleur-CAPE #95: Contrôle de toit ouvrant

LoupHC, 2019-04-06:
Salut Yoan, j'ai fait une petite modif de rien sur le PCB ce matin.
Je cherchais une façon d'intégrer des connexions pour actionner des relais externes plutôt que directement les moteurs.
J'ai réalisé que la sortie moteur agit comme deux sorties 24V qui alternent entre le GND et +24V. Si on connecte chacune de ces sorties à des relais externes, qui eux sont connectés au GND, on obtient deux signaux indépendants. Quand le circuit est censé faire tourner le moteur d'un bord, une sortie est à +24V, donc active, et l'autre est à GND, donc inactive. Quand on change le sens de rotation, les états s'inverse.
Penses-tu que ça devrait marcher?

yoangp, 2019-05-06:
Allo Loup!!
Pour le toit ouvrant, ça peut potentiellement marcher pour inverser le sens du moteur, mais ça prend aussi un état OFF, donc pas le choix que d'utiliser 2 sorties de relais minimum pour le toit ouvrant. Mais surement possible, on a 3 relais extra; on peut en prendre 2 pour le toit ouvrant et 1 pour une fournaise dans ce cas.

LoupHC, 2019-05-10:
Quand le signal n'est ni open, ni close, la sortie moteur peut être considérée à off non?

yoangp, 2019-05-11:
Ok, je reformule autrement, est-ce que tu veux dire qu'on pourrait utiliser 2 relais extra pour le toit ouvrant; le premier ayant les état OFF ou OPEN et l'autre OFF ou CLOSE?

LoupHC, 2019-05-11:
non, mon idée serait d'utiliser la sortie moteur comme sortie pour toit ouvrant, vu qu'elle oscille entre trois états, OFF(circuit ouvert), Polarité1 (24V-GND) et Polarité2(GND-24V). Si on rajoute un troisième contact GND on peut avoir l'équivalent de deux sorties 24V (en les branchant comme ça : M1-GND et M2-GND) On s'appelle cet aprem?