LoupHC/controleur-CAPE #96: PCB - Règles de routing

yoangp, 2019-05-07:
Pour dimensions reliées au routing, commencer par copier ce qui a été fait pour Otomate V1, sinon; spec générales pour routing:

Largeur track puissance 24V:
2mm

Bus d'alimentation 5V, 12V, 24V:
2.5mm si possible, sinon 2mm

Largeur track actionnement des relais, etc.:
1.27mm (anciemment 1mm)

Largeur track de signaux sensors, IO Arduino, low power, etc: 0.635mm (0,254 pour signaux pas long entre pads serrés de l'Arduino par exemple)

Distance minimale entre tracks de signaux: 0.6mm (ref sur internet: 0.4mm)

Distance miminale entre tracks de puissance: 1.0mm

Distance minimale entre pads: 0.6mm

**Pads standards**
Épaisseur du pad à ajouter à la grosseur du trou pour avoir le diamètre du pad :
1.2mm

**Pads de puissance:**
Épaisseur minimale du pad à ajouter à la grosseur du trou pour avoir le diamètre du pad:
1.6mm

**Trous petites pièces through hole:**
Jeu de plus que la grosseur de la patte de la composante (grosseur de la composante = dia de la patte ronde ou diagonale d'une patte rectangulaire, prendre aussi la tolérance maxd de la pin pour les calculs de grosseur de trou):
0.3mm total

**Via standards:**
-Largeur pad: 1.0mm
-Grosseur trou: 0.6mm

**Via signaux:**
Largeur pad: 0.508mm
Grosseur trou: 0.3302mm

yoangp, 2019-05-07:
Loup, t'as utilisé dans méchand petits via sur l'Otomate V1. C'est ou que tu as trouvé la référence que c'était assez gros??

LoupHC, 2019-05-10:
je me rappelle pas d'avoir vu de référence pour la grosseur des vias... je crois que jai pris ceux par défaut...
est-ce que le via doit être plus gros que la trace?

yoangp, 2019-05-22:
[image]