LoupHC/controleur-CAPE #99: Test du nouveau PCB

LoupHC, 2019-07-20:
Après avoir geeké autour du nouveau circuit, j'ai trouvé quelques petits bugs mineurs (rien qui compromette beaucoup le circuit) : 

- les sondes de courant sont affectées par le bruit des relais (c'est des capteurs à l’effet hall), après avoir gossé un peu, avec, j'ai déterminé qu'elle devraient être relocalisées à au moins 5cm des bobines.
- inversion des lignes SDA/SCL pour la chip mcp23008
- petite erreur dans les inscriptions sur les connecteurs : +5V correspond à l'entrée analogique de la station météo et AD3 correspond plutôt à un ground protégé par une résistance 1K
- le pullup de la l'entrée analogique de la station météo (R6) doit être enlevée, ça fausse les données du capteur
- le circuit d'alarme est à refaire (même la dernière version que j'avais envoyé sur github marche pas finalement), je pourrai faire les modifs ça changera pas grand chose au routing.
- la PTC de protection des optocouplers sert pas à grand chose vu qu'ils sont déjà protégés par des résistance 10K
- le gros condo C1 sert-il à quelque chose finalement? Est-ce qu'on peut ajouter des diodes pour l'isoler des moteurs?

J'ai quelques suggestions d'améliorations aussi : 
- ajouter une sortie relais en parallèle des relais on/off moteurs (pour pouvoir convertir les sorties moteurs en sorties devices)
- mettre une ptc 0,5A en série avec l’alimentation des sorties 24V actionnées par les relais 6-7-8, en cas de court-circuit, ça évitera de faire un spark de 6A
- mettre l’alimentation 12V en parallèle de l’alimentation par batterie 9v, avec diodes pour empêcher les retours de courants (comme ça, sauf en cas de panne d’électricité, c’est le power supply 12v qui alimente l’alarme
- ajouter un +gros condos du côté de l’alim 5v (pour coupures de courant brèves)
- je rajouterais une alimentation 12v sur le gros connecteur pour les sondes meteo, ça éviterait d'avoir à mettre deux fils dans le même trou!

Mais bon c'est pas grand chose... une petite soirée de touch-ups et on est all set  je dirais!
Bravo Yoan!!!

jayserf, 2019-07-31:
THAT'S awesome news guys!!!

When is the party for touch ups? I'd like to bring the beers 😁

Congratulations to both of you

LoupHC, 2019-07-31:
Est-ce que tu viens à l'Expo-champ le 1er septembre? 
Le 17 aout on pensait monter un demo de serre automatisée pour montrer là-bas, you should join us!!

jayserf, 2019-08-01:
Great I'll be there! (for both but def on the 17)

Let me know where
And I can laser cut some plastic or wood (1/8 plywood or thicker acrylic)
if we need parts for our mini greenhouse

LoupHC, 2019-08-14:
Après avoir installée l'Otomate 2.0 dans une vrai de vrai serre aujourd'hui, j'ai quelques commentaires aditionnels : 
- les connecteurs des 3 devices sont pas dans le même ordre que les switchs qui les overrident, c'est un peu mêlant (C'est device1-device2-device3 pour les switchs et device1-device3-device2 pour les connecteurs)
- le connecteur jack pour la brancher la batterie 9v est trop prêt de la paroi de la boîte pour mettre une plug jack ordinaire. J'ai réussi à brancher un connecteur jack à angle droit par contre mais on doit le brancher avant de visser le pcb.
- c'est difficile de déplugger les connecteurs du côté droit, on réussit à peine à glisser ses doigts entre les relais et la boîte. Une fois que le cable est installé par contre c'est facile de juste tirer sur le cable pour le déplugger.

Mais malgré ces petits détails, à date, c'est 100% fonctionnel!!!

jayserf, 2019-08-14:
Amazing!

What's up for Saturday?
Where do I meet you two?

LoupHC, 2019-08-15:
Hey man!
I can't anymore for saturday. Besoin urgent d'aller faire mon lavage en ville hahaha. 
I chat on the phone earlier with Yoan we tought maybe we could meet up next saturday instead? (the 24) Does it fit for you?
We could meet at Croque-saison.
Testing the device, making the demo for Expo-champ, discussing lasercut acrylic cover, connectivity issues and electromagnetic noise... bref, some awesome geek-to-geek catch up! 
Sounds nice to me!

jayserf, 2019-08-15:
Hey! Cool for me for this weekend, gives me more time to work on some more
nerd stuff to show you.

What about sunday? (or even this sunday.?)
Sat is the Garlic festival at our market and I dont really want to miss it
(and I should help out too), but again we only have a week

sounds good for the discussions, lots to cover and make.

weird to be making plans on github :-)

jayserf, 2019-08-16:
So no go for this Sunday.

Gotta deliver some roquette to lufa now 😉

LoupHC, 2019-08-16:
Hey it could finally happen this saturday! I changed my plans a bit, i'll be at Lingwick now all weekend. Is it still ok for both of you?
Next weekend it could only be saturday for me tho.
Sorry for the last minut call!

jayserf, 2019-08-16:
Sure, I still can, in lingwick?
Let me know so I can go to sleep early 😉

yoangp, 2019-08-17:
> email_source=notifications&email_token=AGL6IJ3YEJL2SW36GDNTFKLQE4Q6BA
> 5CNFSM4IFPOLKKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5

Ok, ça peut marcher pour moi aussi, sauf que je serai là un peu plus tard,
il faut que je prépare mon matériel à apporter. Je peux être là vers 10h00
à 10h30 à Croque-saisons.

Est-ce que vous pouvez envoyer une confirmation finale?

À demain!

Yoan

LoupHC, 2019-08-17:
Awesome!
Je suggère qu'on se retrouve chez moi au 46 route 108, Lingwick vers 10h30 pour que vous m'aidiez à amener mon stock hehe. La ferme est à 2 minutes de char!

jayserf, 2019-08-17:
Cool! I just got to figure out a car 🤔

See you there!

jayserf, 2019-08-17:
Hey!  for some reason I can't find your phone numbers but I'm in East
Angus. And Google says I'll be there in 20 minutes. My number is
514-781-1122