LoupHC/controleur-CAPE #100: Augmenter la durée de vie des relais-moteurs

LoupHC, 2019-07-21:
Est-ce que ça pourrait être une façon de réduire le sparking?
https://www.illinoiscapacitor.com/pdf/Papers/spark_suppression.pdf

yoangp, 2019-07-23:
> Assigned #100 <https://github.com/LoupHC/controleur-CAPE/issues/100> to
> @yoangp <https://github.com/yoangp>.

J'ai fait un petit calcul rapide et je me trompe peut-être avec la formule
mentionnée dans le document, mais ça me donne une super grosse valeur de
condensateur, genre C = I*I/10 = 5*5/10 = 2.5...   2.5Farad??? Si oui,
c'est malade, mais possible et vraiment gros et vraiment cher. En tout cas,
j'ai pas encore trop approfondi l'affaire. Il faut que j'étudie les
circuits à mon travail pour voir comment ils font pour avoir une durée de
vie de 10ans..

yoangp, 2019-07-28:
Je pense que j'ai une solution intéressante:

1. Modifier le code pour que le fonctionnement des moteurs soit:
1.1 Actionner le relai d'inversion en premier
1.2 Délai
1.3 Actionner le relai d'activation.

Si on peut faire que les délais s'appliquent même à la configuration de l'Otomate V1, alors ce changement a un impact minimal pour les Otomates V1 tout en gardant une seule version du code. Pas d'impact si les relais s'activent pas en même temps sur l'Otomate V1. Il faudrait juste voir ce qui va se passer dans la logique de diodes des drivers de moteurs 2018..

2.Ensuite, on change les relais d'activation de l'Otomate V2 par: https://www.digikey.ca/product-detail/en/ixys-integrated-circuits-division/CPC1709J/CLA280-ND/1277135

LoupHC, 2019-07-28:
Pour la modique somme de 156$, on peut même les avoir sur les même sockets qu'avant... ;)
https://www.digikey.ca/product-detail/en/omron-automation-and-safety/G9H-210S-DC24/Z9531-ND/3475018
L'idée est bonne, et probablement beaucoup plus durable. Par contre je ne suis pas sur comment on peut rétrofitter cette logique-là sur les contrôleurs de l'ancienne version?

LoupHC, 2019-07-28:
Si on passe au solid state, faut aussi penser à des stratégies de limitation de courant, vu que c'est la raison la plus probable de défaillance des relais...

yoangp, 2019-07-28:
Pour la limitation de courant, vu que le power supply est limité à 6,5A, on est pas pire en partant..

LoupHC, 2019-07-29:
Bon point!
Est-ce que le moteur peut générer des spikes de courant lui-même?

yoangp, 2019-07-29:
Possible que le moteur génère des spikes... J'ai commandé 4 ssr, 2 pour moi et 2 pour toi si ça marche.. Le courant peak est de 40A. Courant max coutinu sans heat sink de 9A.

LoupHC, 2019-07-31:
Version chinoise de relais SSR sur sockets à 15$/unit. Compatibles avec les sockets de relais précédents :
https://www.aliexpress.com/item/32883085495.html?spm=a2g0o.productlist.0.0.128bd0b3sjlHOI&algo_pvid=6c85a5e6-5103-48f5-9de4-2d908d1c8e5d&algo_expid=6c85a5e6-5103-48f5-9de4-2d908d1c8e5d-9&btsid=629ec698-5898-4eb4-8f40-1c0150d4ea86&ws_ab_test=searchweb0_0%2Csearchweb201602_1%2Csearchweb201603_52
Pour avoir un délais entre le relais direction et le relais on/off, un circuit comme celui-ci pourrait faire l'affaire  : 
https://www.electroschematics.com/4826/delay-on-circuit/
La séquence d'activation ressemblerait à : 
1. signal envoyé au driver de moteur (active ou non le relais de direction)
2. délais de 1 ms
3. activation du relais on/off

La séquence à l'arrêt serait  : 
1. arrêt instantané du SSR
2. le relais de direction relâche quelques microsecondes après, le temps que la bobine se décharge

Dans les deux cas le ssr gère l'inrush et l'interruption de courant, le changement d'étât du relais de direction se fait "à vide"

jayserf, 2019-07-31:
Are you guys having problems with relay life in your greenhouses?

Ours are still working super well, have not had to change one yet.

or is this for moving to solid state?

Sorry im late to the conversation , catching up now.

LoupHC, 2019-07-31:
Some had trouble with sparking in the direction relay... causing bad contact and motor opening instead of closing...
SSR would be a great solution to this issue... plus theres a chinese module with the same form factor than previous relays!

jayserf, 2019-08-01:
Got it. Thanks good for the ssr and keeps the design compatible with cheap
mechanical relays also.

Don't they need heatsink sometimes though? Or we're at too low current and
voltage?

C$ 7.91  5%OFF | SSR-40 DA-H DC-AC Solid State Relay SSR 40A 3-32V DC /
90-480V AC w Heat Sink
https://s.click.aliexpress.com/e/bkebzT3S

yoangp, 2019-08-01:
> email_source=notifications&email_token=AGL6IJ3PCHPLCXD3IVN3KV3QCIMUVA
> 5CNFSM4IFS2QO2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5

Ya, at low current, some ssr don‘t require heatsink.

LoupHC, 2019-08-01:
Under 5A they rarely need one

jayserf, 2019-08-04:
That's sweet! Good idea then, thanks

LoupHC, 2019-08-18:
Hey guys!
J'ai eu la possibilité de tester le nouveau circuit de driver de moteur ce matin! Ça marche au quart de tour!
Le condensateur prend environ 1 seconde à se charger et 1 seconde à se décharger, ce qui est en masse suffisant pour supprimer le problème de sparking.
Aussi les SSR sont plugs and play avec l'ancien driver de moteur (pour le relais on/off) ça peut donc être une solution partielle à ceux qui ont des problèmes de sparking...

Concernant ton inquiétude Yoan d'avoir une activation partielle vu que la led semblait s'allumer graduellement, j'ai testé ça à l'oscilloscope. Le ssr a probablement effectivement un mécanisme de déclanchement abrupte parceque ça semblait assez instantané, de l'ordre de quelques ms.

yoangp, 2019-08-19:
Cool! Maintenant ça serait un bon test de comparer ce setup avec le setup
de relais actuel via un cyclage automatique pour voir combien de cycles ça
peut marcher et convertir cela en années d’utilisation.. Apres l’expo-champ.

Yoan

LoupHC, 2019-08-19:
Nice!
Faudra juste se trouver une pesée de 100kg, après j'ai la poulie qui faut! :P

yoangp, 2019-08-19:
Allo Loup,

Est-ce que tu peux m’envoyer un petit plan de ton circuit de délai et aussi
le plan du circuit d’alarme aisni que les numéro de pièces de tout cela?

Je vais commencer à réviser le circuit..

Merci

jayserf, 2019-08-19:
You guys are MACHINES!

My question, if we add the delay circuit components and still use a
mechanical relay, does it reduce the sparking at all?

yoangp, 2019-08-20:
Hey Jason!

So the idea is for one motor we use one solid state relay (SSR) and one
mechanical relay. The delay circuit first close the mechanical relay,
without load (so no spark), and after the delay it activate the SSR. No
spark in the SSR under load.

Ciao!

Yoan

yoangp, 2019-08-20:
Loup,

Ça me fait penser, il faut que le circuti de relai fonctionne pour
démarrage et arrêt. C'est à dire qu'il faut activer le SSR après le relai
mécanique pour le démarrage du moteur. Mais pour l'arrêt, il faut d'abord
désactiver le SSR, puis le relai mécanique.

Est-ce que le circuit de délai tient compte de cela?

Sinon, je sais que t'aime pas l'option software, mais j'ai pensé qu'il
pourrait y avoir une pin de l'arduino shorté à 5V et lorsque le code
détecte cela, il sait qu'il s'agit du circuit de l'Otomate V2 et alors il
introduit le délai software.

C'est quand même pas si grave de changer des relais sur l'Otomate V1 une
fois de temps en temps. En plus, c'est pas tout le monde qui aura
rapidement des problèmes, Jason par exemple n'a eu encore aucun problème de
relai et Reid m'a dit qu'en général il n'a pas trop de problèmes de ce côté
là non plus.

jayserf, 2019-08-20:
OK I get it, so doing the modif to theboard and still using mechanical
donesnt change anything.

i know you explained it like 8 times sat...thanks

well I still maintain that the relays are so cheap and easy to find that
unless we do testing to show its a problem thats widespread its more effort
and cost than just having spare relays and changing them once a year like
batteries in a thermostat.

is it possible that with the new programming where only one rollup works at
a time, then problems are really minimized and it might be years before
some fail?
or does it really depend on meteorological conditions etc than it does on
load?

yoangp, 2019-08-20:
> email_source=notifications&email_token=AIGEN7FQ7IKMOCVQWOWQ7UTQFJVC3A
> WW2ZLOORPWSZGOD4SLOLQ#issuecomment-522499886
> AIGEN7AZGEGD3RWZC525DSLQFJVC3ANCNFSM4IFS2QOQ
> email_source=notifications&email_token=AGL6IJYCISECATQKZ4TUIYDQFMFSXA
> WW2ZLOORPWSZGOD4UK6HA#issuecomment-522759964
> AGL6IJYADFZV2UHAVU7YO4DQFMFSXANCNFSM4IFS2QOQ
> email_source=notifications&email_token=AIGEN7GEFDHXZPWEZF6YXILQFMN5PA
> WW2ZLOORPWSZGOD4UP3ZI#issuecomment-522780133
> AIGEN7FF5NZUUB5SVK7KA6LQFMN5PANCNFSM4IFS2QOQ
> email_source=notifications&email_token=AGL6IJYSX5GK7D4Z674CALTQFNATNA

Hi jason, goos question, but one motor at a time doesn’t help with this
problem.

The problem with failing relay is mainly when it happen, you may have a big
problem if the roll up are stuck because of a failed relay when it’s
freezing outside or crazy hot inside. So a higher fiability is a big plus!

Ciao

jayserf, 2019-08-20:
OK! agreed, didnt see that point, its true, and worth the effort.

so solid state lasts forever right :-)

and can be swapped with a regular relay anytime, so you can make the space
for 2 by spacing out the relays a bit?

LoupHC, 2019-08-20:
Le circuit en question!
À l'arrêt le SSR s'arrête assez instantanément, la charge maximale du condo est assez proche de la valeur de la zener donc pas trop de charge résiduelle pour garder le transistor ouvert. Sinon on peut augmenter la valeur de la résistance de décharge. Faudrait checker ça avec un oscilloscope 2 voies si on veut être bien sûr...
[image]

LoupHC, 2019-08-20:
A failed inversion relay makes the rollups going straight up! Not so nice during automn nights!

yoangp, 2019-08-20:
La logique pourrait aussi être inversée pour qu'un relais d'inversion kaput
fait que les roll-ups descendent. Ça aide quand il fait froid, mais pas
quand il faut chaud!

yoangp, 2019-08-20:
C'est quoi le numéro du transistor de délai?

LoupHC, 2019-08-20:
s9014 je crois...
ouen je pense que j'aime mieux que ça ouvre par défaut, un peu de froid ça se tolère (la majorité du temps dans l'année), une serre qui monte à 50 ça pardonne pas.... peu importe le moment de l'année.. ça serait mon bet

yoangp, 2019-08-20:
Ouais, avec-vous lu le message de Yan, 45degrésC dans sa serre qui s'est
pas ouverte... J'ai écrit à Yan pour savoir si c'est à cause d'un bug ou
défectuosité d'une de nos versions de boîtier de puissance/contrôleur.

La fiabilité de notre système devrait être un must!

LoupHC, 2019-08-21:
Tout compte fait le circuit de driver déconnecte le relais du GND, donc la charge restante du condensateur devient instantanément la seule alimentation du relais solid state, du relais d'inversion et  du transistor qui maintient le tout en fonction. ça doit se passer extrèmement vite.
Une façon de diminuer le délai de relâchement serait de diviser la valeur du condo, et de doubler d'autant la valeur des résistances de charge.