LoupHC/controleur-CAPE #101: Proto V2.0 - Interférences mesure de courant (ASC723)

yoangp, 2019-07-29:
Tout ce qu'il y a à savoir d'après moi:
https://www.allegromicro.com/en/insights-and-innovations/technical-documents/hall-effect-sensor-ic-publications/managing-external-magnetic-field-interference-acs71x-current-sensor-ics

Donc voici les 3 solutions que je vois:
1. On éloigne les ACS723 des relais
2. On trouve une façon de faire tenir le ACS723 verticalement pour réduire le flux magnétique des relais qui le traverse. (Possible mais trop cher il me semble: https://www.digikey.ca/product-detail/en/aries-electronics/08-810-90/A117-ND/106399)
3. On met un shield comme mentionné dans l'article ci-dessus. Sur Digikey ils sont des tapes qui peuvent faire un shield, mais j'ai pas encore trouvé un shield pour soic8.

LoupHC, 2019-08-09:
J'ai essayé de faire un shield maison avec des assiettes en allu... mais à date je suis pas convaincu...
J'ai aussi essayé l'inclinaison perpendiculaire, je pense que l'interférence de la bobine est juste trop forte pour tout ça, ce qui marche vraiment, c'est de déplacer les capteurs de courant, à au moins 5 cm.
Pas la solution la plus élégante, mais au moins la plus facile pour avoir à changer tout le routing du pcb, on pourrait mettre des sockets à l'autre extrémité du pcb, et les connecter au circuit de relais avec un flat cable comme ça : 
https://www.digikey.ca/product-detail/en/samtec-inc/IDSS-16-D-05.00-G/SAM9041-ND/4386575
Faudrait juste faire les calculs pour voir combien de cable ça prend pour supporter le courant des moteurs (le ribbon cable en question c'est du awg28)

yoangp, 2019-08-09:
> IDSS-16-D-05.00-G/SAM9041-ND/4386575

Allo Loup,

Alu ça marchera pas, il faut un matériau magnétique comme le fer qui va
dévier le magnétisme. J‘avais pensé à un tuyau de plomberie en acier coupé
en 2 sur la longueur qui couvre les 2 asc723 comme un tunnel.

Sinon, à voir si on déplace les asc723 plus loin sur le pcb, c‘est surement
faisable.

Pas sur pour le flat ribbon,  c‘est fait pour des signaux. Je pense que ça
en prendrait vraiment beaucoup en parallèle.

À plus,

Yoan

LoupHC, 2019-08-09:
Si on en utilise 3 en parallèle pour chaque input de moteur ( 3 in + 3 out) on pourrait théoriquement monter jusqu'à 4A en restant dans les specs du fil, on aurait besoin de 10 connecteurs par sonde de courant donc un connecteur de 2x10 pourrait faire l'affaire pour les deux sondes.
Mais bon, c'est à utiliser au pire des cas...
Je vais essayer avec un tuyau morceau de tuyau en fer!
http://www.mosaic-industries.com/embedded-systems/electronic-instrument-design-new-product-development/cables/ribbon-cable-current-rating

LoupHC, 2019-08-31:
Tout compte fait, j'ai trouvé que la solution la plus simple consistait à mettre une plate de métal 1-8 entre les relais et le reste du board. La lecture des sondes de courant passe de 0.2 à 0.05A quand le courant ne circule pas dans les moteurs mais que les relais à proximité sont allumés.

yoangp, 2019-09-02:
Bon à savoir! Ça fait un shielding suffisament efficace...

Ciao!

Yoan