LoupHC/controleur-CAPE #102: Proto V2.0 - LCD interférences

yoangp, 2019-07-30:
De https://community.particle.io/t/i2c-lcd-display-getting-corrupted-solved/9767/28:

À essayer:
-Voir s'il y a des pull-ups en parallèle (sur le circuit et back-pack du LCD), semble être le cas. Enlever les pull-ups du circuit sur SDA et SCL pour voir si ça aide.

-"I2C depends strongly on timing. So normaly, during a I2C communication every interrupt should be disabled / suppressed (including Serial Communication)"...

-"Add some 0.01uF and 100uF capacitors across the power supply leads to the display as close to the display as possible."

-"Make every wire as short as possible.", peut-être flt-ribbon du lcd de 2in plutot que 4in si possible à trouver.

-Update de librairie pour le LCD...

yoangp, 2019-07-30:
Allo, enlève les pull-up R8 et R10 sur le I2C, ça aide déjà beaucoup l'écran.

LoupHC, 2019-07-31:
Tu constates déjà une différence??

yoangp, 2019-07-31:
Oui!

LoupHC, 2019-07-31:
Quelle genre de différence?

jayserf, 2019-07-31:
That sucks...

there's always the 27,000 cables on digikey to choose from for shorter ones
though, 1 problem solved.

https://www.digikey.ca/products/en/cable-assemblies/rectangular-cable-assemblies/450?FV=ffe001c2&quantity=0&ColumnSort=1000011&page=1&k=ribbon+cable&pageSize=25

yoangp, 2019-07-31:
Hello Jason,

No problem here, in fact it helps that I removed the pull-ups on the I2C data lines (R8 and R10) because there is already pull-ups on the LCD backpack I2C pcb. I believe it helps because voltage value for "0" is closer to 0V. No big change here on the Otomate PCB, just no R8 and R10 resistors installed or removed altogether from the next PCB design.

jayserf, 2019-08-01:
Ok cool that we're not having flickering screen issues. Tks