LoupHC/controleur-CAPE #103: Proto V2.0 - Modif rain software

yoangp, 2019-08-04:
Allo! :-)

Modification apportés au code pour que le rain bucket du Sparkfun fonctionne:

- Ajustement du paramètre de quantité de pluie pour le tipping bucket
Dans Grennhouse_3.sensors.h, modification de la quantité de pluie par tipping du bucket:
_#define Bucket_Size **0.2794** // bucket size to trigger tip count **(mm of rain)**_

- Modification de GetRain pour que l'activation rain = true et rain = false fonctione correctement (voir section RG11 en gras que j'ai utilisé pour le Sparkfun):
_void getRain(){
  //Normally open contact rain sensor
  if(greenhouse.rainSensor.value() == NO_CONTACT){
    if(digitalRead(RAIN_SWITCH) == LOW){
      rain = true;
    }
    else{
      rain = false;
    }
  }
  //Normally close contact rain sensor
  else if(greenhouse.rainSensor.value() == NC_CONTACT){
    if(digitalRead(RAIN_SWITCH) == HIGH){
      rain = true;
    }
    else{
      rain = false;
    }
  }
  **//RG11 - 0.01'' bucket rain sensor
  else if(greenhouse.rainSensor.value() == RG11_BUCKET){
    //bucket mode
    if(totalRainfall > rainSample + rainSetpoint){   //If rainfall increased 
      if(rainSampleCounter < rainSampleDelay){   //If the time since the last tipping bucket is shorter that the max delay between two tipping of the bucket, then rain started
        rain = true;
        rainSample = totalRainfall;      //Ajust current rainfall total based on actual total rainfall
        rainSampleCounter = 0;           //Reset counter to time till next rainfall increase
      }
      else{
        rainSampleCounter = 0;            //Reset counter to time till next rainfall increase
        rainSample = totalRainfall;       //Ajust current rainfall total based on actual total rainfall
      }
    }
    else{                 //No new bucket tipping since last time GetRain looped
      if(rain = true){     //Rain already True
        if(rainSampleCounter > rainSampleDelay){   //If the time since the last tipping bucket is longer that the max delay between two tipping of the bucket, then rain stopped
          rain = false;
        }
      }
    }
  }**
}_

- Ajout de commentaires et valeurs utilisées pour mes tests:
_float rainSetpoint = 0.5; //Min mm of rain to decide that it's really raining
unsigned long rainSampleDelay = 10000; //Max delay between two rain bucket tipping that define that it's raining (normalement pour déclancher  à pluie modérée de 2.5mm/h : 720000 = 12minutes)_

- Ça prendrait aussi un paramètre sur l'interface pour ajuster ce paramètre quand il faut que la variable rain s'active en dixième de mm/hr. (ex sur l'affichage: "Min rain trigger: [2.5] mm/h")

LoupHC, 2019-08-04:
Normalement le compteur prend seulement le compte quand le signal de High à Low, donc un signal bloqué à low va seulement être compté comme un click.
Merci pour les commentaires! Je vais checker ça pour programmer le rain bucket dans l'interface

yoangp, 2019-08-04:
En fait, j'ai aussi oublié, si le code ou les paramètres sont
nécessairement défférents du RG11, est-ce que tu peux ajouter dans le menu
de config des capteurs météo : NC, NO, RG11 rain bucket* ou Sparkfun rain
bucket*.

Quand tu auras le temps...

Merci!