LoupHC/controleur-CAPE #104: Expo-champs 2019 - setup de démo

yoangp, 2019-08-04:
Allo,

Loup, est-ce que tu peux voir s'il y aurait un morceau de plastique de serre chez croques saisons qu'on peut utiliser pour la mini-serre pour l'expo champ (et nos futurs tests après...)?

Je verrais une mini-serre en forme de prime à base triangulaire équilatéral, chaque côtés du triangle de 6pieds de long (pour utiliser les tiges de support à moteurs que j'ai déjà sans les modifier) et longueur de la mini-serre, genre 6 pieds aussi.

Donc, il faudrait un morceau de plastique rectangulaire de 13pied par 6 pieds et 2 petits morceaux en triangles pour fermer les bouts. Ça peut aussi être une bâche ou du plastiques d'ensillage.

Je pense que j'ai les planches 2x4 qu'on pourrait utiliser pour la structure.

LoupHC, 2019-08-05:
13'x6'!!!
Faut pas oublier que ça doit rentrer dans mon char hahah!
Sinon j'ai des retailles de galvanisé chez nous, au lieu de prendre le tiens on pourrait plutôt faire un mini bras articulé de rollup (genre 2 pieds de long, pour pas trop avoir ça dans les jambes à l'expo-champ).
Je me disais qu'on pourrait fixer ça après le board de démo, pour servir d'équerre au board en même temps, donc 3 pieds de long max.
C'est sûr que ça serait moins impressionnant comme installation mais... un peu moins intense à assembler et trimballer? :P
Je pense pouvoir piquer une retail de wirelock et le plastique pour l'installation!

yoangp, 2019-08-05:
Ok, à voir pour la grandeur. Je pensais à un montage facilement
montable/démontable avec quelques vis.

Liste de matériel:
-2x4 yoan
-Plastique de serre Loup
-Vis à bois Yoan
-Support à station météo Yoan
-2ieme prototype Yoan
-Station météo - Yoan/Loup
-Moteurs - Yoan
-Chaufrette - Yoan
-Ventilateur carré 120V - Loup? J'en ai un d'ordi 12V avec power supply...
mais c'est assez poche
-Power supply - Yoan/Loup
-Perceuse et mèches - Yoan
-Scie sauteuse - Yoan
-Ruban à mesuer - Yoan
-Kit de soudure & électronique - Yoan/Loup
-Capteurs temp/hum/lumière- Yoan/Loup
-Multimètre - Yoan/Loup