LoupHC/controleur-CAPE #106: Proto V2.0 Capteur SHT10

yoangp, 2019-08-09:
Allo Loup,

Je viens de recevoir le capteur SHT10 que j'ai commandé sur Amazon. Hé ben, c'est exactement le même que celui que j'ai commandé d'Aliexpress, il vient assurément de la même compagnie chinoise. Dans tous les cas, j'ai encore le même problème, aucune communication.

Est-ce que tu pourrais regarder le tiens et le séparer à la base du capteur et me dire à quoi correspond les numéros de pins 1 à 4 (Vcc, GND, DATA, CLOCK). Peut-être que le code de couleur des cables n'est pas bon...
[image]

Aussi quand j'ouvre le capuchon métalique de mon capteur, je peux voir que les pins 1 et 2 sont du côté top layer du pcb du capeur, le côté sur lequel est soudé le capteur d'humidité. Mais je peux rien voir de plus car tout est dans un genre de rubber imperméable.

Merci!

LoupHC, 2019-08-13:
Peut-être que ça peut t'aider, j'en ai branché une tantôt et donc j'ai dû me demander quel fil va où...
Dans mon cas : rouge (5V) bleu (gnd) jaune et noir (clock/data me rappelle plus quel est quel)

Entre le 5v et le gnd, sur le pcb de la sonde, il y a un petit condo de filtrage. J'espère qu'il est pas complètement enfouie dans le coating dont tu parles.

yoangp, 2019-08-14:
Moi j'ai pas de fils rouge, j'ai un fils brun dans les 2 cas, mais mon fils brun est supposément le 5V.. Il y a peut-être juste de quoi pas correct dans mon code, on va voir cela en fin de semaine.