LoupHC/controleur-CAPE #107: Proto V2.0 - Wind

yoangp, 2019-08-09:
Allo!

J'ai testé le capteur de vent avec le kit sparkfun, ça marche :-) Woohoo!!

Il manque cependant quelque chose dans le code Loup. On peut mettre la vitesse du vent qui déclanche la fermeture des côtés ouvrants, mais il faut aussi mettre un délai avant la réouverture de la serre car lors d'une tempête de vent ça vient par bourasques.

Voici l'algorythme proposé:
-Dans le menu, après que l'usager à sélectionné la vitesse du vent maximale, il doit entré la durée du délai de réouverture.

Dans le code, il faut que ce délai corresponde à un countdown. À chaque fois que la vitesse du vent redépasse le seuil de fermeture défini pas l'usager, il faut que le countdown se reset.

Note sur le code:
Loup, est-ce que c'est ok pour toi si je te dis les ajouts de codes qu'il faut mettre pour l'instant vu que c'est toi le mastermind derrière tout cela, tu as une meilleure vue d'ensemble. Si tu préfères qu'on ajoute du code chacun de notre bord pour te sauver du travail, il faut s'entendre pour que le dernier code soit toujours à jour sur Google drive et qu'on s'avertisse par Hangout quand quelqu'un travailles sur le code.

Sinon, pour l'instant, voici ce que j'ai ajouté pour que je puisse utiliser mon anémomètre à switch (car le tiens était analogique):
Dans Greenhouse_3.sensors.h:

_#define switch_delay_to_wind_speed 2.4 // 2.4km/h = 1 trigger/second_
_volatile unsigned long contactTimeAne; // Timer to manage any contact bounce in interrupt routine for anemometer_ 
_elapsedMillis AneSampleCounter;    //Anemometer switch counter; elapsed time in millisecond since last trigger_

_// Interrupt handler routine that is triggered when the anemometer make one rotation
void isr_ane() {
  if((millis() - contactTimeAne) > 15 ) { // debounce of sensor signal
    windSpeed = switch_delay_to_wind_speed*1000/AneSampleCounter;
    if(windSpeed < 1){
      windSpeed = 0; //Avoid low speed when no wind, so if windspeed lower than 1km/h than reset to 0km/h
    }
    AneSampleCounter = 0;
    contactTimeAne = millis();
  }
}_

_void getWind(){
  //  Yoan commented analog anamometer
  //0-5V (0-30 m/s) analog wind sensor
//  int aRead = analogRead(WIND_ANALOG_SENSOR);   
//  windSpeed = (float)108/(float)1023*(float)aRead; //108km/h = 30m/s
}_

Il faudrait aussi ajouter du code pour sélectionner l'anémomètre à switch ou analogique dans le menu.

yoangp, 2019-08-09:
By the way, ça prendrait aussi un raccourci dans le menu pour accéder à l'écran "Weather station", je pense qu'il y a juste via Config/sensors/weather/... qu'il est possible de voir cet écran. C'est bien cool de regarder ces données météo quand on pris dans la serre en pleine tempête!! :-)

yoangp, 2019-08-09:
Hey, je suis en pleine grosse tempête juste là là!! Plate que mes capteurs météo de l'Otomate soient dans la maison et pas dehors.. Bah, si le toit arrache, je vais pouvoir voir des données météo malade!!

LoupHC, 2019-08-10:
Hey malade! Nice job pour le programme d'interrupt avec l'anémomètre. Je vais pouvoir l'intégrer facilement.
On va aussi pouvoir le rendre compatible avec le capteur de Davis instruments avec ĉa!
Je pensais à ça justement, ajouter un délai avant la reconfirmation de la vitesse du vent. L'autre paramètre qu'on peut ajuster, c'est l'hystéresis avant de désactiver l'override de fermeture.
Mis tout ensemble ça ferait : 
1. surveillance en temps réel de la vitesse du vent
2. Si la vitesse dépasse X km/h, on ferme et on part un timer
3. Toute les Y minutes, on vérifie si la vitesse du vent est passée en dessous de X km/h - hysteresis
pour savoir si on doit réouvrir, sinon, on reset le timer

LoupHC, 2019-08-10:
Je pense que c'est bon si tu me copy paste tes modifications, pour que je les assimile avant de les intégrer au code officiel.
C'est rendu tellement organique mon affaire... des fois je dois remonter 4 ou 5 pages de code pour dépister des bugs, c'est bon que je me tienne à jour! :)

yoangp, 2019-08-11:
Allo!

Je suis pas certain de comprendre à 100% l‘idée de l‘hytérésis. Il faudrait
pas que la serre s‘ouvre parce qu‘il y a eu une mini accalmie entre deux
bourasques dans une tempête de vent..
Est-ce que tu peux m‘expliquer ton idée plus en détails?

Yoan

LoupHC, 2019-08-12:
Exemple, s'il le setpoint de vent est à 70km/hm et que l'hysteresis est de 10, ça veut dire qu'il va pas ré-ouvrir avant que le vent redescende en bas de 60km/h.
Mais en y pensant bien, pas sûr que ça soit la meilleure solution. Faudrait plutôt se baser sur une moyenne des 10 dernières minutes mettons.

yoangp, 2019-08-12:
Par rapport à ton idée de moyenne, je dirais que c‘est les bourasques qui
peuvent arracher une serre. Mettons une tempête à 60km/h de moyenne sur
10min, mais avec des pointes à 90km/h, c‘est ça qui me ferait plus peur..
Mais l’idée d‘hystérésis pourrait être bonne afin d‘ouvrir la serre quand
la vitesse de vent a réellement baissé à un niveau acceptable. Je vote pour
cela.

Yoan