LoupHC/controleur-CAPE #108: PCB V2.1 - Espace

yoangp, 2019-08-20:
Allo,

Donc j'ai pensé au PCB seconde version. Voici les changements que je propose:

- On garde la même boîte et la même grosseur de PCB, on est capable ;-)
- On fait de l'espace entre les connecteurs pour les doigts.
- On met un connecteur barrel jack vertical et le circuit d'alarme dans le bas du circuit pour que la batterie soit simplement dans le fond de la boîte.
- On met le module RTC et SMD card vertical pour sauver de l'espace. (le module RTC toujours dans le bas de la boîte pour éviter les problèmes de batteries du module RTC qui coulent)
- On met des fusibles plus petits pour sauver de l'espace.
- On enlève les trous inutilisés autour de l'Arduino.
- On met les ACS723 en SMD.
- On ajoute un thermistor pour en avoir un par moteur.

à suivre

jayserf, 2019-08-20:
> Assigned #108 <https://github.com/LoupHC/controleur-CAPE/issues/108> to
> @jayserf <https://github.com/jayserf>.

I know its a slippery slope, but surface mount Varistor and thermistor? I
didnt know they existed:

https://www.digikey.ca/products/en?keywords=VARISTOR%2047V%202KA%20

really easy to desolder guys! I promise

Is it bad to not remove the arduio pin expansion? I think its really useful
for the like 5 or 10 of us that would add other shit to the arduino...

yoangp, 2019-08-20:
> email_source=notifications&email_token=AGL6IJZFV5EO6CQFUL5WIU3QFNCGDA
> 5CNFSM4INK4CCKYY3PNVWWK3TUL52HS4DFWZEXG43VMVCXMZLOORHG65DJMZ

Hi Jason,

I would say, for the ones who want to add extra stuff on the Arduino, they
can buy a shield. And seriously, who would develop other stuff from scratch
on the Arduino? Maybe Jenna, Loup, Reid, you and me. Who would do it, less
than that. And us 3 doesn’t count because we will built a new circuit
instead..

But I can try to leave a footprint for I2C and SPI communication.

Is that a good compromise? :-)

Ciao,

Yoan

LoupHC, 2019-08-20:
I actually need them for the wifi relay interface... i could select a few and tell you which tho... and we remove the extras.
Mais à part ce détail je suis 100% d'accord avec tes propositions Yoan! Bonne idée de mettre les modules à la verticale!

yoangp, 2019-08-20:
Ok super, tu me diras quels io garder accessibles pour ton module relay
wifi. :-)

jayserf, 2019-08-20:
Hey Yoan.

Yeah agree with you on that one.

Not sure about upside-down shield use, might be complex but possible...

Ok for spi and i2c (I only need spi)

Thanks!

yoangp, 2019-08-27:
> Après délibérations avec moi-même, j'ai décidé que les actions
> télécommandables à distance seraient :
> Pour les trois rollups : ouvert-50%-fermé-auto
> Pour les 5 devices : on-off-auto
> Pour changer la météo à distance : soleil-mimi-nuage
> J'ai donc besoin d'avoir accès à 30 ios inutilisés, plus un accès au
> ground.
> Si tu me laisses les ios 14-21, A8-A14 et au moins une des deux rangées de
> pins dans le bas du MEGA, je vais être correct pour tout ça!

Ok! :-)

jayserf, 2019-08-27:
😎